tetrabiodistributed/papra-pcba #14: Buzzer when battery is about to run out

mmroden, 2021-02-08:
Feature request: when the battery is low, an audible signal as well as the gauge would be helpful

mmroden, 2021-02-13:
version 0.4 at the earliest

mechatroniks-git, 2021-04-23:
Will add this buzzer to v0.5
12V Buzzer - https://lcsc.com/product-detail/Buzzers_Fengming-YS-MBZ12095DYB12_C409840.html/?href=jlc-SMT
[datasheet](https://datasheet.lcsc.com/szlcsc/1912171631_Fengming-YS-MBZ12095DYB12_C409840.pdf)

mechatroniks-git, 2021-05-01:
Buzzer sourced locally: https://www.jameco.com/z/CEM-1212C-CUI-Electromagnetic-Audio-Buzzer-12V-35mA-85dB-2-3kHz_2286415.html
 https://www.jameco.com/Jameco/Products/ProdDS/2286415.pdf

mmroden, 2021-05-22:
closed by #49

closed 2021-06-01 by mechatroniks-git

mechatroniks-git, 2021-07-21:
Digikey Buzzer: https://www.digikey.com/en/products/detail/intervox/BRT1209P-12-C/13693933

PR #52 (2021-06-01): V.5 Release candidate
Closes #14
Closes #19
Closes #36
Closes #46
Closes #47
Closes #48