tetrabiodistributed/papra-pcba #15: Papra PCB engineering drawing

mmroden, 2021-02-10:
Please submit an engineering diagram in PDF, PNG, whatever makes sense.

Bonus points for a soldering jig.

darraghbr, 2021-02-10:
[A-TET- Papra Control PCBA Drawing.pdf](https://github.com/tetrabiodistributed/PAPRA-PCB/files/5961311/A-TET-.Papra.Control.PCBA.Drawing.pdf)

@mechatroniks-git Please review when able.

mechatroniks-git, 2021-02-11:
Looks nice. A few comments:
1. The board, as it comes from JLC should have the fuse installed, no need to point that out.
2. The PCB should have a balloon
3. Washers & Nuts included with the potentiometer and DC jacks - not sure if they should be separate line items?
4. Lightpipe is already on the M12 controller BOM? It should only be on 1 BOM, probably the other one, as it needs to be assembled with the board, after the board is installed.

[image]

darraghbr, 2021-02-11:
Please see attached, I believe I have dealt with all of your comments. 

Aside from the nuts and washers appearing on the BOM, Because they are seperate components there isn't a nice way to not have them visible on the BOM and also have them visible on the drawing. I can remove them from the drawing and BOM if desired. 

[D-TET-.Papra.PCBA.Drawing.pdf](https://github.com/tetrabiodistributed/PAPRA-PCB/files/5968976/D-TET-.Papra.PCBA.Drawing.pdf)

mechatroniks-git, 2021-02-11:
LGTM

darraghbr, 2021-02-12:
@mmroden please review when able.

mmroden, 2021-02-12:
Man, this is _phenomenal_!

closed 2021-02-12 by mmroden