tetrabiodistributed/papra-pcba #17: Papra fan box PCB engineering drawing

darraghbr, 2021-02-12:
Submit an engineering diagram for the Papra fan box PCB in PDF

darraghbr, 2021-02-12:
First draft for review by @mechatroniks-git 

[D-TET-001101-0 PapraFanBoxPCBA Drawing.pdf](https://github.com/tetrabiodistributed/PAPRA-PCB/files/5971070/D-TET-001101-0.PapraFanBoxPCBA.Drawing.pdf)

mmroden, 2021-02-13:
Version 0, not with the new airtight connectors

mechatroniks-git, 2021-02-15:
This version looks good. 

The sealed connector PCB, v0.3 has been committed, ready for review

darraghbr, 2021-02-20:
Updated for new DC jack and general tidy up.

[D-TET-001101-0 Papra Fan Box PCBA Drawing.pdf](https://github.com/tetrabiodistributed/PAPRA-PCB/files/6016248/D-TET-001101-0.Papra.Fan.Box.PCBA.Drawing.pdf)

mechatroniks-git, 2021-02-20:
Looks good, but maybe something moved in assembly? 
[image]

darraghbr, 2021-02-20:
The previous version of the drawing also showed a gap between the components and the board? If you want me to fix it I will.

mmroden, 2021-02-22:
Closed per discussion, work has been completed as there is a drawing for the papra pcba

closed 2021-02-22 by mmroden