tetrabiodistributed/papra-pcba #19: Move PAPRa firmware to designated repo

capsulecorplab, 2021-02-14:
Personally, I would move the PAPRa firmware to a designated repo, as to provide clearer association of the codebase to its license and separation of version history between PCB & firmware code. The firmware repo can then be added as a submodule, if the PCB repo must reference the firmware - keeping in mind that submodules need to be updated.

mechatroniks-git, 2021-04-26:
@capsulecorplab Do I need to do this in a manner that preserves history, or just create a new Repo and copy the files over and remove them from here?

capsulecorplab, 2021-05-01:
@mechatroniks-git I would probably do the latter, then add replace the firmware folder in this repo with a submodule that points to the firmware repo. https://github.blog/2016-02-01-working-with-submodules/

mechatroniks-git, 2021-05-15:
https://github.com/tetrabiodistributed/papra-pcb-firmware created.

mmroden, 2021-05-22:
closed by #49

PR #52 (2021-06-01): V.5 Release candidate
Closes #14
Closes #19
Closes #36
Closes #46
Closes #47
Closes #48
closed 2021-06-01 by mechatroniks-git