tetrabiodistributed/papra-pcba #21: Resolve the exact model of fan

darraghbr, 2021-02-17:
After chatting with Darryl it seems there is a mismatch between the fans that are specced and the standard fans that are the same form factor. 

With some napkin mathematics I do not believe that the fan that has been specced (from amazon) is delivering the flow rates and pressures that are quoted, as ultimately it is drawing the same wattage but is somehow delivering significantly higher performance than other standard fans.

darraghbr, 2021-02-18:
If we purchase a standard model like this one:

https://www.digikey.co.uk/product-detail/en/nmb-technologies-corporation/BG1203-B055-000-00/P12911-ND/740268

Then do some comparative tests between the fan acquired from amazon.

mmroden, 2021-02-22:
Testing involves a multimeter to measure current draw, to make sure that they blow the same amount of air.  Checking to see if the flow is about the same (for some given value of "the same")

darraghbr, 2021-02-22:
Updated my comment with the correct model of fan.

mechatroniks-git, 2021-04-23:
Tested the following blowers:
- [NMB BG1203-B045-00S-00](https://www.digikey.com/en/products/detail/nmb-technologies-corporation/BG1203-B045-00S-00/857938?s=N4IgTCBcDaIAoEYDMAWFBWAtAOQCIgF0BfIA)
- [Sunon PMB1212PLB2-A.(2).GN](https://www.digikey.com/en/products/detail/sunon-fans/PMB1212PLB2-A-2-GN/1021212?s=N4IgTCBcDa4KwE4C0BGAzANjUgcgERAF0BfIA)

mmroden, 2021-04-24:
We will go with Sunon.  This will increase the BOM by $10 per individual part to $25/fan.  The amazon one can be recommended as an alternate source item for lower cost.

closed 2021-04-24 by mmroden