tetrabiodistributed/papra-pcba #25: What ATTiny ICs are compatible

darraghbr, 2021-02-24:
There are dozens of different ATTiny ICs out there, we have specified the 1604 on the PCB, is that the minimum required? Are there any other SKUs that are compatible?

darraghbr, 2021-02-24:
So in theory any of these should work as long as the program takes less than 2 KiB of flash storage, 128 bytes of RAM and 64 bytes of EEPROM. Is there any way we can verify this?

ATtiny204, ATtiny404, ATtiny804, ATtiny1604 
ATtiny214, ATtiny414, ATtiny814, ATtiny1614

mechatroniks-git, 2021-02-24:
I would say that any ATtiny__4SSN/ATtiny__4SSNR would work.  
- 4SS is 14 pin SOIC device
- N is temp range with 20MHz. We could do the F, and get to 16MHz, but would have to adjust firmware for clock change (baud rate, pwm rate, blink rate, etc)
- R is just a packaging type for manufacturing (R for tape & reel), blank for tube or tray

The 0 series ATtiny_04, I've tested 204, 404 and 1604. This series is more than capable for this application and is lowest cost. 
- 204 has limited flash, would require serial debug to be disabled and may not have room for future updates, but can be supported for now. May not support any future features (for example, we activate the expansion header for RF integration).
- The 404 is recommended for price vs current and future capabilities. 
- The 804 should drop in without any concern.  
- The 1604 tested and works, only installed it due to JLC availability (only chip they had at the time of order)

The 1 series ATtiny_14 chips have more SRAM and more hardware IO, but pin mapping is unchanged and should be compatible. The extra cost is not justified for this application, at this time. Code compiles for these devices without error, but should be tested if needed for compatibility confirmation. 

[image]

mechatroniks-git, 2021-02-24:
Appnote on migration from 0 Series to 1 Series:
http://ww1.microchip.com/downloads/en/AppNotes/Microchip%20AN2636%20Migration%20Between%20tinyAVR%201-Series%20and%20tinyAVR%200-Series%2000002636A.pdf

confirms migration from 0 to 1 without any code changes
[image]

darraghbr, 2021-02-24:
Great work Patrick, I've noticed on JLC at the moment they have had some shortages of various ATTiny ICs and we have had to go with the 1614 on the boards for 3DCrowd. Perhaps some point down the line a list of compatible SKUs kept somewhere in the docs?

mechatroniks-git, 2021-02-24:
We can say the following are compatible:
*ATtiny204, *ATtiny404, ATtiny804, *ATtiny1604
ATtiny214, ATtiny414, ATtiny814, ATtiny1614
Ones with * being tested/confirmed

mechatroniks-git, 2021-02-24:
Code already has the following comment
```
  PCB is compatible with the following 14SOIC ATtiny chips: 404/804/1604 and 414/814/1614 should work (needs testing)
  Attiny204 - this could work without serial output and some code optimization
```
I'll add the SSN clarification on next commit

closed 2021-02-24 by mechatroniks-git