tetrabiodistributed/papra-pcba #29: BOM scrub

mmroden, 2021-03-01:
Need to check to make sure that the BOM is updated, and order necessary parts.

Order fans through digikey

mechatroniks-git, 2021-03-05:
I built a DigiKey Cart for the PAPR components I need. See https://www.digikey.com/short/r80dwtpq

@DHHwangPhD Can you add in the blowers you wish to test?

mmroden, 2021-03-05:
requires amazon blower as well as the digikey fan so that we can compare

mechatroniks-git, 2021-03-06:
I added a couple of blower candidates to the cart using a dart board and blind fold selection process.
[image]

mechatroniks-git, 2021-03-08:
New cart: https://www.digikey.com/short/2jnh3hmq

darraghbr, 2021-03-08:
That NMB SKU is a PWM 3 wire fan, I assume you don't want that?

mechatroniks-git, 2021-03-08:
Final Cart https://www.digikey.com/short/wd4tffcn

mechatroniks-git, 2021-03-08:
And will need 1 of these for crimping: https://www.amazon.com/IWISS-Terminal-Crimping-SN-01BM-0-08-0-5mm/dp/B019ARWWFY

closed 2021-03-08 by mechatroniks-git