tetrabiodistributed/papra-pcba #34: waterproof/splashproof the papra?  How to do that?

mmroden, 2021-03-11:


darraghbr, 2021-03-12:
We could use a set of louvres on the fan box to hopefully keep the worst of any rain etc off the filter pack. 

It may limit which orientation you can mount the fan box.

darraghbr, 2021-03-12:
For the M12 controlled we could use a 3D printed TPU or other flexible gasket that seals between the battery and the M12 housing. The lid could then be attached permanently with a waterproof adhesive and IP rated potentiometers and DC jacks utilised.

darraghbr, 2021-03-12:
An example of the louvers is something like what are used on armoured fighting vehicles to protect the radiators and air intakes. 

They are also used on dirt bikes to protect the radiator from debris. 

[image]

mmroden, 2021-03-20:
Handled by #37 and #38 in this project

mmroden, 2021-08-11:
redundant to #55 and work done in the papr-print-files

closed 2021-08-11 by mmroden