tetrabiodistributed/papra-pcba #36: Bad fan boards can cook the m12 controller boards

mechatroniks-git, 2021-03-14:
When testing a "bad" fan board, multiple m12 v2 controller boards and a Milwaukee m12 charger were all rendered nonfunctional once power was applied to the system when testing a fan.

mmroden, 2021-03-20:
Should be solved by the 0.3 board

mmroden, 2021-03-29:
Now will be solved by not having a fan board

mechatroniks-git, 2021-03-29:
Possible bad wiring in fan box (dead short on fan power wiring or in fan itself) can cause same issue. PCB V3+ should protect.

mechatroniks-git, 2021-04-17:
Added 1A fuse on 12V to DC blower, tested with dead short on fan output, board survived and is fully functional once short is removed.  Recommend adding 2nd fuse to DC blower on v.5 PCB, so board should have 2 fuses in parallel, one for DC blower, one for everything else.

[image]

mechatroniks-git, 2021-04-17:
[image]

[image]

mmroden, 2021-05-13:
Closed by #43

mmroden, 2021-05-22:
closed by #49

PR #52 (2021-06-01): V.5 Release candidate
Closes #14
Closes #19
Closes #36
Closes #46
Closes #47
Closes #48
closed 2021-06-01 by mechatroniks-git