tetrabiodistributed/papra-pcba #41: Investigate blown rev .2 PCB

mechatroniks-git, 2021-04-17:
At the end of the 4/10/22 build party, tested new IP rated DC cable, which had short, fried Mark's M12 control board.

mechatroniks-git, 2021-04-17:
Powered up m12 board, found 5V regulator was hot and only outputting 1.4 VDC.

mechatroniks-git, 2021-04-17:
MOSFET also blown. Replaced, board is now working. Same failure mode, shorted output can fry the MOSFET and ATTINY. Will be resolved in V.5 of PCB, for now, don't short the output and you will be good.

ATtiny was replaced with 214, but 2K code space fills up, so had to remove startup knight-rider LED sequence to fit. Same issue likely with the 204 flavor. Also, v2.3.1 of the MegaTinyCore has a bug with this chip, created issue. V2.9 works.

darraghbr, 2021-04-24:
It might be worth adding an eFuse IC? It would replace the polyfuse, and could provide overcurrent, overvoltage and short protection.

https://www.ti.com/power-management/power-switches/efuse-hotswap-controllers/overview.html?keyMatch=EFUSE&tisearch=search-everything

mmroden, 2021-05-22:
closed by #43

closed 2021-05-22 by mmroden