tetrabiodistributed/papra-pcba #54: LED's not flashing green

capsulecorplab, 2021-07-02:
This may be a non-issue, since it happened after using it to power a stepper motor driver for a NEMA17, but the LED arrays don't flash green when the control knob is turned back on and instead goes directly to red. The battery seems fine and I don't see any burnt IC's. Any thoughts?
[image]
[image]
[image]
[image]

mmroden, 2021-07-02:
Iirc the LEDs are all one color, not multiple colors for different states.  Feature request, maybe?

capsulecorplab, 2021-07-02:
@mmroden you're right, the LED color was red. but this issues is concerning the LED arrays that indicate the battery level that aren't coming on

mechatroniks-git, 2021-07-19:
@capsulecorplab - v.2 had a critical failure mode that was fixed later. Issue was if the output fan was shorted, the microcontroller would blow up. 

If you connected a low impedance load (NEMA stepper motor) to the FAN output, this would likely cause the chip failure.

mmroden, 2021-08-14:
@mechatroniks-git will get @capsulecorplab a replacement