tetrabiodistributed/papra-pcba #55: coat the m12 board for water proofing

mmroden, 2021-07-17:
mask up board after it's built, spray with some spray can conformal coating.  Will require documentation, of course

mechatroniks-git, 2021-07-17:
Acrylic is easiest to apply, least toxic and makes for easy rework. It has decent moisture protection. Silicone would be a good step up.

This acrylic looks like a good option: https://www.digikey.com/en/products/detail/acl-staticide-inc/8690/10270032

Conformal coatings will be easily removable with MG8310A coating stripper. https://www.mgchemicals.com/downloads/tds/tds-8310a-l.pdf

Background reading: https://www.techspray.com/the-essential-guide-to-conformal-coating

darraghbr, 2021-07-17:
Could also fill the case with potting compound but this would make any maintenance down the line much more challenging.