tetrabiodistributed/papra-pcba #59: e-Fuse chip tripping at mid-range

mechatroniks-git, 2021-08-29:
eFuse is holding at max current (2.6A) with 100% duty cycle. In the mid range, duty cycle is under 100%, eFuse will trip, unsure why. Tried adding caps (1st was a 10uF tantalum, then a 50uF electrolytic). Both seemed to help, suspect dv/dt. Was testing with power supply, possible better outcome with a battery, but none were charged.

darraghbr, 2021-08-31:
Just to clarify did you add caps to the voltage input or in seies with Cdvdt?

darraghbr, 2021-08-31:
Page 26 of the data sheet here. It states the fastest slew rate can be achieved by leaving the dVdt pin floating, might be worth a try? 

Might be worth trying to measure the load capacitance and inrush current (although I'm sure we discussed the load capacitance and agreed it would be negligible?)

[image]

mechatroniks-git, 2021-08-31:
@darraghbr - I added the caps between the 12V rail & ground, after the fuse.

We're powering up OK (most of the time), just during run time, when the blower is running at mid-range duty cycle, we're tripping, so my theory is the DV/DT is the cause. I'll remove that dv/dt cap and test later today. Thanks for looking that up.

mechatroniks-git, 2021-09-01:
I removed C5, still tripping. We're seeing lots of noise on the battery and fused output. These are images right above the trip point, at duty cycle of ~66% ON. 

Not sure what the problem is, ringing or if it's the ringing in the OFF or ON, but more filtering will help. I need to study the datasheet further, but if DVDT is inrush limiting, we would benefit from having a big (470uF?) electrolytic cap after the fuse to prevent that droop. 

[image]

[image]

mechatroniks-git, 2021-09-01:
Here's a better grab, ch 1 & 2 at same amplitude, 1/2V per division. The battery voltage (ch2) not moving much.

Concerns: 

- When fet turns off, we get ringing and a back EMF spike. We have a flyback diode, but it may not be fast enough.  

- The voltage drop across the fuse is substantial, big droop at turn on

[image]

mmroden, 2021-09-01:
What do the waveforms look like with the known-good board?  Are we sure these waveforms are indicative of the underlying problem?  (not saying they shouldn't be resolved, but more that we're fixing the right thing)

darraghbr, 2021-09-02:
What does the PG pin read during these events?

ultraapa, 2021-09-07:
I think I understand the problem better now that I'm reading this post. The decoupling caps are there for initial high dv/dt voltage spike situations. The flyback protects against reverse current by a collapsed magnetic field from low or no voltage situations. So I would not think the caps and flyback would resolve the problem as the problem is at run time when I would think the voltage spikes are not present and voltage is good I hope. The decoupling and flyback (should be mandatory) are good design practices in any event. That being said, on initial blush it sounds like there may be some kind of "runaway" situation where current is being driven higher and higher until it causes a trip situation at the fuse. . I wonder if the the chip (TPS25947???) providing the current to the fuse is overheating? I have some questions but I'll those for in person discussion. It would interesting; however, to know the status of pin PG as Darragh suggests.

mechatroniks-git, 2021-11-07:
Comparing the eval board to our schematic:

1. TVS Diode on power input
2. 1uF cap on power input
3. R1 & R3 = 1M & 124K, while board has 470K & 75K. This sets the undervoltage lockout threshold @ 10.87V. Ours is set to 9V to match the battery discharge curve. 
4. 100pF cap in parallel with R3
5. PGTH R2 & R4 set to 1M & 124K, while board has 360K & 56K (PG = 9V)
6. 2 Output caps @ 10uF and 220uF, while board has only 1uF
7. Schottky diode on output for protection

[image]

mechatroniks-git, 2021-11-07:
DV/DT cap had been removed in prior testing, left it off.
Adding a 560uF 50V cap (what I have on hand) provided the fix.  Board seems to be working fine with this addition.

[image]

Here is output to the fan @ 50% duty cycle
[image]

mechatroniks-git, 2021-11-07:
Still need to test board for:

- Reverse polarity protection 
- Undervoltage and overvoltage protection
- Test current monitoring with ILM (needed for CFM confirmation)

Recommended changes for V.7.
1. Add 220uF AL cap to output of fuse
2. Change C7 to 10uF
3. Add 100nF in parallel with R15
4. Add 1uF to input
5. Add schottky diode on the output

PR #63 (2022-06-02): Updating to v0.7
Closes #59
commit ac1bbe62 (2022-06-02): Updating to v0.7

Closes #59
closed 2022-06-02 by capsulecorplab