tetrabiodistributed/papra-pcba #67: Investigate replacing the e-fuse with a toshiba part

mmroden, 2022-08-13:
Efuses are hard to find, we need a replacement part.  This ticket is an investigation for whether this part can replace the existing efuse.

https://www.digikey.co.uk/en/products/detail/toshiba-semiconductor-and-storage/TCKE800NL-RF/12324957

mechatroniks-git, 2022-08-15:
Quick comparison of features:

[image]

##TLDR:

### Toshiba
- Toshiba part will be easier to solder and should have less components required
- Toshiba requires external MOSFET for reverse polarity protection

### TI
- TI part is cheaper and more featured
- TI has adjustable Overvoltage, Toshiba has a fixed value @ 15.1V
- TI has slightly lower on-resistance (less heat) - but still good
- TI has status outputs
- TI has better nuisance tripping resilience
 
[image]

mechatroniks-git, 2022-08-15:
[image]

https://octopart.com/tcke812na%2Crf-toshiba-109207110?r=sp
vs
https://octopart.com/tps259472arpwr-texas+instruments-113046828?r=sp

mechatroniks-git, 2022-08-15:
TI circuit:

[image]

Toshiba Circuit:
[image]

darraghbr, 2022-08-19:
Just a couple of quick agenda items for the meeting.

- What is the stock like for the transient suppression diodes and the external FET
- Do we currently take advantage of the reverse polarity protection on the current board rev (I seem to recall that we do) and do we think that it is necessary to keep that functionality (I think it is)

mechatroniks-git, 2022-08-19:
Reverse polarity is a good thing, certainly worth the $.31 per board.

These are the recommended parts, all made by Toshiba. Cost at 100 board qty.  Lots of stock at the moment. 
TVS: Lots of stock, $0.20/board [octopart](https://octopart.com/df2s23p2ctc%2Cl3f-toshiba-87695657?r=sp)
SBD: Lots of stock, $0.22/board [octopart](https://octopart.com/cuhs20s30%2Ch3f-toshiba-97210178?r=sp)
FET: Lots of stock, $0.31/board [octopart](https://octopart.com/ssm6k513nu%2Clf-toshiba-75194178?r=sp)

darraghbr, 2022-08-20:
https://www.digikey.co.uk/en/products/detail/toshiba-semiconductor-and-storage/TCKE812NA-RF/12324959?s=N4IgTCBcDaICoGEDSBRAHARjAOQIIgF0BfIA

Updated digikey link for the correct over voltage and stock

mechatroniks-git, 2022-12-22:
With Texas Instruments showing stock, request was made to order 250 parts for stock to avoid having to design in the Toshiba part. 

Closing this issue for the time being.

closed 2022-12-22 by mechatroniks-git